phrack/ShootOFF #686: NPE on roundTargets.getIterator in steel challenge when shooting a target after recalibration

phrack, 2017-03-16:
Pops here with an NPE:

https://github.com/phrack/ShootOFF/blob/master/src/main/java/com/shootoff/plugins/SteelChallenge.java#L234

My guess is the exercise isn't being restarted after recalibration.

Calibrate, add target, calibrate, shoot target should dupe it.

cbdmaul, 2017-03-16:
Is this one of those cases where shot detection is broken because something threw an NPE in its thread?

Regardless, seems like we should catch UEs at the shot thread level and warn the user.  Preferably send us the log too, breaking shot detection should be a priority bug

phrack, 2017-03-16:
There is a different issue we need a seperate thread for. The shot detection thread passes new shots to the canvas manager and then to the training exercises. This means anything that happens in a `shotDetected` thread in an exercise happens on a shot detection thread. In theory this does not slow things down since one shot detection thread = one shot. However, this does cause two problems:

1. An exception on the `shotDetected` method can break shot detection.
2. Training exercises unnecessarily occupy a shot detection thread that would otherwise immediately return to the pool. 

I'll open a separate issue to deal with this.

commit 32beace3 (2017-03-18): Fix shot detected NPE in steel challenge to close #686
closed 2017-03-18 by phrack