phrack/ShootOFF #691: Selecting a target without distance information without perspective manager causes warning popup

cbdmaul, 2017-03-19:
It complains about there not being perspective information (i.e. pm is not present) but the target doesn't have perspective tags anyway.  Warning is useless and confusing.

phrack, 2017-03-19:
I am actually a bit conflicted in this case because it's not actually useless (maybe confusing though). If there are not defaults set in the target file you can still fill in the blank boxes in the target distance panel and it will work. For example, with the POI target if I fill in the width and height of the target and the distance behind the projection screen it does resize in a reasonable way if I chance the distance after filling in initial values. 

Another way to say that is because the panel lets you override the defaults you can also set defaults if there were none.

cbdmaul, 2017-03-19:
But if there is no perspective manager then changing those does nothing?

phrack, 2017-03-19:
The warning is telling you what to do so you can have a perspective manager. If you don't have a perspective manager it won't even show you the distance panel.

cbdmaul, 2017-03-19:
Even with that intent it is a horribly misguided message.  It is presented as a warning about missing "critical" data.  the perspective support is a super cool additional feature not something it absolutely needs to work correctly.  the message reads like a complaint not a great extra thing shootoff can do.

phrack, 2017-03-19:
I don't know what to change it to. You're the one that wanted the warning in the first place to guide people to use the paper. I am and always was perfectly happy to just silently not use PM. I get the green banner thingy if it worked and I'll also get the distance pane so I already have more than enough feedback to know that I can use distances.

cbdmaul, 2017-03-20:
I suggest we change it to a more friendly "ShootOFF supports real world sizes! But you're not set up for it, blah blah blah" AND only show it once (ever) like the first start dialog.

cbdmaul, 2017-03-20:
In addition, I suggest adding a banner for not finding the perspective paper pattern so that people that do have it have feedback on whether or not it was discovered.  That way everyone had a chance to know what it means (the one time dialog) and can get feedback from the banner that isn't obtrusive.

commit 37d0301c (2017-03-20): Rephrase perspective manager missing message as a positive and only show once per install to close #691
closed 2017-03-20 by phrack