phrack/ShootOFF #693: Target doesn't appear correctly if set to arena size

cbdmaul, 2017-03-19:
https://github.com/phrack/ShootOFF/blob/3fef7fbe927424f38cc7514ff8f5820585167a26/src/main/java/com/shootoff/gui/pane/ProjectorArenaPane.java#L602

This code:

target.setPosition(0, 0);
target.setDimensions(arenaStage.getWidth(), arenaStage.getHeight());

Results in:

2017-03-19 11:35:14 [JavaFX Application Thread] DEBUG c.s.gui.pane.ProjectorArenaPane - target Point2D [x = 0.0, y = 0.0] - Dimension2D [width = 1280.0, height = 720.0]

But the target is off the top left of the arena

phrack, 2017-03-19:
Ugh, when I dump every position value I can think of it shows the expected values for everything but it's blatantly not in the right place.

phrack, 2017-03-19:
OK, I got most of it. It's because the projector arena pane is an anchor pane and if you dump `getTranslateX()` and `getTranslateY()` when full screening they are both negative numbers that are exactly how off the target is. What I can't explain at the moment is where these negative numbers come from. In the constructor they are both 0.

cbdmaul, 2017-03-23:
[image]

Still seeing this bug I think? That's what the arena looks like after adding the poi target.  It's off the top of the arena.

phrack, 2017-03-24:
I'll see if this dupes in my VM, but this might be a high resolution problem because me and erpesjo are having the correct outcome:

https://www.reddit.com/r/ShootOFF/comments/60jwd8/its_that_time_again_please_help_us_prepare_for/df914mc/

cbdmaul, 2017-03-24:
My arena is only 800x600, isn't that low res compared to yours?

Main display is 4K but that doesn't have the arena on it

phrack, 2017-03-24:
I don't know what JavaFX does in this case. It looks like a coordinate problem not a sizing problem and I assume the main window is on the high resolution display. This seems to leave room open to be the resolution issue.

phrack, 2017-03-25:
It has something to do with the aspect ratio. If I don't use a projector and just manually resize the arena on my main screen I get a very similar outcome to your screenshot with some aspect ratios but it works just fine at others.

phrack, 2017-03-25:
Easiest way to dupe is to open the arena without an extra screen and drop it down to make it tall and skinny. Target always seems to be the right size, just not in the right place.

commit 1fdbd4de (2017-03-19): Use anchor pane translation to adjust target position to close #693
commit 62e05015 (2017-03-25): Fix #693 by converting origin coordinates from local to parent
closed 2017-03-25 by phrack