phrack/ShootOFF #697: Shot placement not agreeing across feed, arena, and arena tab

cbdmaul, 2017-03-27:
Not sure when this bug developed.  I reviewed the code and I don't see anything obvious.

In "detect in bounds only", the shot is placed in the right place on the arena tab.  It is forced into a small square (the bounds?) in the arena and is off by bounds from left and top in the feed.

In "detect everywhere", the arena placement is still wrong but the feed is correct.

Maybe we should subclass Shot into ShotForFeed, ShotForArena, ShotForArenaTab to force a type exception if our logic is wrong?

cbdmaul, 2017-03-28:
The problem is that the marker is not being set except for in setTranslation.

I entirely refactored Shot before realizing that.  Now I'm not sure which way to go...refactor is mostly working and passes all the tests but it touches a lot of code.

phrack, 2017-03-28:
I am not entirely clear based on the details here. The Javadocs are clear on translate but the practice isn't. Layout coords are supposed to be the permanent position and translate coords are meant for temporary offsets (e.g., for animations). Generally translate offsets aren't supposed to show up when you view the node's coords. Weird things happen in practice.

In any case, here is my shot at what I'll want to test for shot refactoring:

- [ ] Session recorder with shots on empty canvas
- [ ] Session recorder with shots on animated region (e.g. a plate)
- [ ] Session recorder with shots on scored target with Shoot for Score
- [ ] Shots on animated target on arena
 
If those didn't break, nothing else should have.

cbdmaul, 2017-03-28:
Tests are passing and some critical thinking leads me to believe that everything should work the same (Since the class is the same but just subclassed a bunch).  Everything worked when I tested but I didn't get to set up my projector to test everything.

Lets rip the bandaid off now, I'll test tonight and push it up.

closed 2017-03-29 by cbdmaul

Erpesjo, 2017-03-29:
Works for me. Perfect! Good job!

phrack, 2017-03-30:
I ticked off the boxes I had above with no problems too.

commit 6fd6d980 (2017-03-29): Fixes #697 with a refactor of the Shot class