phrack/ShootOFF #698: Clicking save in preferences too rapidly causes all cameras to be locked open

cbdmaul, 2017-04-04:
Also causes exceptions (Probably in already open cams?):

Exception in thread "FetchImageCellWebcamImages" java.lang.UnsupportedOperationException: Provided data element number (0) should be multiple of the Mat channels count (1)
	at org.opencv.core.Mat.put(Mat.java:2550)
	at com.shootoff.camera.cameratypes.OptiTrackCamera.translateCameraArrayToMat(OptiTrackCamera.java:179)
	at com.shootoff.camera.cameratypes.OptiTrackCamera.getFrame(OptiTrackCamera.java:187)
	at com.shootoff.camera.cameratypes.OptiTrackCamera.getBufferedImage(OptiTrackCamera.java:194)
	at com.shootoff.gui.CheckableImageListCell.fetchWebcamImage(CheckableImageListCell.java:271)
	at com.shootoff.gui.CheckableImageListCell.lambda$1(CheckableImageListCell.java:165)
	at java.lang.Thread.run(Thread.java:745)

cbdmaul, 2017-04-04:
Not sure about this one.  I get the optitrack error consistently but not the locked cams.  I only noticed it by accidentally double clicking/

phrack, 2017-04-04:
Yea, doesn't dupe at all for me after multiple attemps.

cbdmaul, 2017-04-04:
Let's just close it, if it is an optitrack problem nobody but me will get it anyway.

closed 2017-04-04 by cbdmaul