phrack/ShootOFF #700: full screen arena target is stretched off bottom of arena tab

cbdmaul, 2017-04-04:
[image]

on actual arena it appears correctly.  in arena tab it is stretched.  this is on linux with low res screens.

phrack, 2017-04-04:
The problem is that the arena tab tries to always fill the width of the tab area with the arena while maintaining the same aspect ratio as the real arena. With some aspect ratios this means that you end up with an arena area that fits the width but is taller than the available space. The only thing to do in this case is probably to just add a horizontal scrollbar. The alternatives, which I hate, are don't fill the width (would look awful) or don't maintain the aspect ratio (would make things in the tab look squashed or stretched).

phrack, 2017-04-04:
Adding a scrollbar definitely solves the problem but at the expense of the scrollbar always being there for reasons I don't really get at the moment.

cbdmaul, 2017-04-04:
Can you fit to vertical AND horizontal? Meaning, if width or height is too wide or tall, use that as the maximum and scale the other measurement down.

phrack, 2017-04-04:
That may be the best option. Sometimes you'll have space on the RHS and other times the bottom, but never both at the same time.

Adding this to the top of `ShootOFFController.addNonCameraView` works OK but you end up with way more scroll bar than necessary a lot of the time:

```java
		final ScrollPane s = new ScrollPane(content);
		s.setPrefViewportWidth(cameraTabPane.getPrefWidth());
		s.setPrefViewportHeight(cameraTabPane.getPrefHeight());
		s.setHbarPolicy(ScrollBarPolicy.NEVER);

		final Tab viewTab = new Tab(name, s);
```

phrack, 2017-04-04:
What I am about to commit tries to fit both ways and it works with my projector and when I manually resize the arena without a projector to try different aspect ratios. God I hope it works on your projector because I really don't like dealing with JavaFX scaling.

Close this issue if it works for you.

cbdmaul, 2017-04-05:
[image]

Really close...

phrack, 2017-04-05:
That's so close I'll give it 20 more minutes but I am accepting the bug if I don't find the problem in that time box.

phrack, 2017-04-05:
PS. One reason I am willing to accept this is if it's not a 20 minute fix is that it's not a new bug. With this kind of thing I always checkout an old tag to see if it's a new problem a diff will show us the cause of. This has been around since the arena tab was added. We never noticed because before it was just cutting off the bottom of a background and all of the backgrounds have bottoms where cutting a bit off is hard to notice. It's only obvious now because this one niche target makes it obvious.