phrack/ShootOFF #701: Some exercises don't work due to Shot changes

cbdmaul, 2017-04-05:
Wasn't sure if you were aware of this.  What's going to happen here? If we recompile them all won't they break if anyone adds them to an older version?

Exception in thread "Shot Notifier" java.lang.AbstractMethodError: com.shootoff.plugins.ShootOut.shotListener(Lcom/shootoff/camera/shot/Shot;Ljava/util/Optional;)V
	at com.shootoff.gui.CanvasManager.addArenaShot(CanvasManager.java:695)
	at com.shootoff.gui.MirroredCanvasManager.addArenaShot(MirroredCanvasManager.java:186)
	at com.shootoff.gui.CanvasManager.addShot(CanvasManager.java:638)
	at com.shootoff.camera.shotdetection.ShotDetector.lambda$0(ShotDetector.java:130)
	at java.lang.Thread.run(Thread.java:745)

phrack, 2017-04-05:
Typically we just change the minimum version so that updates and new downloads aren't available on old versions. If anyone wants a new install on a n old version, oh well, go manually download it or upgrade.

I'll look into this to see if there is a way to avoid a breaking change.

phrack, 2017-04-08:
Easiest change is a slightly less optimal organization but my distaste for the slightly less optimal organization is far outweighed by my distaste for recompiling and releasing all of the modular exercises for the time being.

commit 1cadf98c (2017-04-08): Move Shot back to camera package to fix #701
closed 2017-04-08 by phrack