phrack/ShootOFF #705: POI Adjustment Question

nszzya, 2017-07-13:
Firing 5 shots at the center crosshair doesn't result in consistent beeps registering and inconsistent loud final beep at the fifth shot. In other words, sometimes I get the high pitched beep, sometimes I get the regular shot on target sound and sometimes nothing. I've had the loud confirmation beep on the second shot and often not at all. 

Also, when finished with the POI Adjustment, how do I remove the target from the display in order to see my shooting target/background?

Thanks for any help.

phrack, 2017-07-13:
You can click the target and then hit the delete key to remove it. The POI adjustment stuff is a target just like any other target.

I have no comment regarding the rest as I didn't write the code. However, a video almost always helps in these kinds of circumstances.

cbdmaul, 2017-07-14:
I'd have to see video, not sure what circumstance would lead to it giving the confirmation beep before the correct number of shots.

nszzya, 2017-07-14:
I’ll post a video.