phrack/ShootOFF #712: Trouble with Manjaro / Arch Linux

jtdaling, 2018-10-09:
Hi,

I am not able to get ShootOFF running on my Manjaro machine.
I ran the ShootOFF diagnostic program and got the following message in my log file:

Runtime: Java(TM) SE Runtime Environment 10.0.2+13 64-bit
OS: Linux 4.14.70-1-MANJARO amd64, Patch Level: unknown
Max amount of memory JVM will use = 1972 MB
Intel(R) Core(TM) i5-5257U CPU @ 2.70GHz
 is64bit = true
 physical cpu(s): 2
 logical cpu(s): 4
CPU Load Per Processor: 14.64% 19.13% 17.94% 17.90%
Total RAM Installed: 7883 MB
Available RAM: 3867 MB
ShootOFF Output:

exit code: 1

stdout:
---

stderr:
---
SLF4J: A number (6) of logging calls during the initialization phase have been intercepted and are
SLF4J: now being replayed. These are subject to the filtering rules of the underlying logging system.
SLF4J: See also http://www.slf4j.org/codes.html#replay
java.lang.reflect.InvocationTargetException
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.base/java.lang.reflect.Method.invoke(Method.java:564)
	at javafx.graphics/com.sun.javafx.application.LauncherImpl.launchApplicationWithArgs(LauncherImpl.java:473)
	at javafx.graphics/com.sun.javafx.application.LauncherImpl.launchApplication(LauncherImpl.java:372)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke0(Native Method)
	at java.base/jdk.internal.reflect.NativeMethodAccessorImpl.invoke(NativeMethodAccessorImpl.java:62)
	at java.base/jdk.internal.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43)
	at java.base/java.lang.reflect.Method.invoke(Method.java:564)
	at java.base/sun.launcher.LauncherHelper$FXHelper.main(LauncherHelper.java:941)
Caused by: java.lang.UnsatisfiedLinkError: no opencv_java2413 in java.library.path: [/usr/java/packages/lib, /usr/lib64, /lib64, /lib, /usr/lib]
	at java.base/java.lang.ClassLoader.loadLibrary(ClassLoader.java:2654)
	at java.base/java.lang.Runtime.loadLibrary0(Runtime.java:876)
	at java.base/java.lang.System.loadLibrary(System.java:1875)
	at nu.pattern.OpenCV$SharedLoader.<init>(OpenCV.java:190)
	at nu.pattern.OpenCV$SharedLoader.<init>(OpenCV.java:185)
	at nu.pattern.OpenCV$SharedLoader$Holder.<clinit>(OpenCV.java:225)
	at nu.pattern.OpenCV$SharedLoader.getInstance(OpenCV.java:229)
	at nu.pattern.OpenCV.loadShared(OpenCV.java:179)
	at com.shootoff.Main.main(Main.java:813)
	... 11 more
Exception running application com.shootoff.Main

Graceful shutdown took too much time, forcing the shutdown.

Regards and thanks for your help!

phrack, 2018-10-11:
This is a known problem with the OpenCV problem we use:

https://github.com/openpnp/opencv/issues/33

Only option at the moment is to use an older version of Java.

jtdaling, 2018-10-12:
Thanks a lot, your solution worked!
For those that use Manjaro or Arch, here is how I got Oracle Java 8 installed:
First I installed yay (Yet Another AUR Helper) since Oracle's Java is not available under Pacman:
```
sudo pacman -S git

git clone https://aur.archlinux.org/yay.git

cd yay

makepkg -si
```
Then I searched for available JDK's with the following command:

`yay jdk`

And I choose the option:

> aur/jdk8 8u181-1 (+58 2.91%) 
> Oracle Java 8 Development Kit

ShootOFF has started, I couldn't check if it runs ok since I don't have a webcam with me to test it.

vonnieda, 2018-11-17:
Hi folks, https://github.com/openpnp/opencv/issues/33 is now fixed so you should be able to upgrade and be back in business.