phrack/ShootOFF #713: Installation on Raspberry Pi 3

jtdaling, 2018-11-08:
Hi!

I've tried to install Shootoff and Java on an Raspberry Pi 3 B+ with Raspbian, LXDE and tightvnc.
Goal is to make a static setup for my projector and camera so that I don't have to go through the whole process of setting up all the hardware each time I want to shoot.

I followed the following directions to install java and all the dependencies:
https://www.reddit.com/r/ShootOFF/comments/5bx9h9/shootoff_on_the_raspberry_pi/

Unfortunately I don't get any further then a message saying:
Error: Could not find or load main class com.shootoff.Main

Shootoff-diagnostics gives the following output:
SLF4J: A number (6) of logging calls during the initialization phase have been intercepted and are
SLF4J: now being replayed. These are subject to the filtering rules of the underlying logging system.
SLF4J: See also http://www.slf4j.org/codes.html#replay
Runtime: Java(TM) SE Runtime Environment 1.8.0_191-b12 32-bit
OS: Linux 4.14.71-v7+ arm, Patch Level: unknown
Max amount of memory JVM will use = 224 MB
ARMv7 Processor rev 4 (v7l)
 is64bit = false
 physical cpu(s): 1
 logical cpu(s): 4
CPU Load Per Processor: 15.25% 18.81% 17.36% 11.30%
Total RAM Installed: 927 MB
Available RAM: 654 MB
ShootOFF Output:

exit code: 1

stdout:
---

stderr:
---
Error: Could not find or load main class com.shootoff.Main

An exception due to the connection occurred, a lockdown will be initiated.
com.getsentry.raven.connection.ConnectionException: An exception occurred while submitting the event to the Sentry server.
        at com.getsentry.raven.connection.HttpConnection.doSend(HttpConnection.java:186)
        at com.getsentry.raven.connection.AbstractConnection.send(AbstractConnection.java:71)
        at com.getsentry.raven.connection.AsyncConnection$EventSubmitter.run(AsyncConnection.java:171)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
        at java.lang.Thread.run(Thread.java:748)
Caused by: java.net.ConnectException: Connection refused (Connection refused)
        at java.net.PlainSocketImpl.socketConnect(Native Method)
        at java.net.AbstractPlainSocketImpl.doConnect(AbstractPlainSocketImpl.java:350)
        at java.net.AbstractPlainSocketImpl.connectToAddress(AbstractPlainSocketImpl.java:206)
        at java.net.AbstractPlainSocketImpl.connect(AbstractPlainSocketImpl.java:188)
        at java.net.SocksSocketImpl.connect(SocksSocketImpl.java:392)
        at java.net.Socket.connect(Socket.java:589)
        at sun.net.NetworkClient.doConnect(NetworkClient.java:175)
        at sun.net.www.http.HttpClient.openServer(HttpClient.java:463)
        at sun.net.www.http.HttpClient.openServer(HttpClient.java:558)
        at sun.net.www.http.HttpClient.<init>(HttpClient.java:242)
        at sun.net.www.http.HttpClient.New(HttpClient.java:339)
        at sun.net.www.http.HttpClient.New(HttpClient.java:357)
        at sun.net.www.protocol.http.HttpURLConnection.getNewHttpClient(HttpURLConnection.java:1220)
        at sun.net.www.protocol.http.HttpURLConnection.plainConnect0(HttpURLConnection.java:1156)
        at sun.net.www.protocol.http.HttpURLConnection.plainConnect(HttpURLConnection.java:1050)
        at sun.net.www.protocol.http.HttpURLConnection.connect(HttpURLConnection.java:984)
        at com.getsentry.raven.connection.HttpConnection.doSend(HttpConnection.java:157)
        ... 5 common frames omitted
An exception occurred while sending the event to Sentry.
com.getsentry.raven.connection.ConnectionException: An exception occurred while submitting the event to the Sentry server.
        at com.getsentry.raven.connection.HttpConnection.doSend(HttpConnection.java:186)
        at com.getsentry.raven.connection.AbstractConnection.send(AbstractConnection.java:71)
        at com.getsentry.raven.connection.AsyncConnection$EventSubmitter.run(AsyncConnection.java:171)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
        at java.lang.Thread.run(Thread.java:748)
Caused by: java.net.ConnectException: Connection refused (Connection refused)
        at java.net.PlainSocketImpl.socketConnect(Native Method)
        at java.net.AbstractPlainSocketImpl.doConnect(AbstractPlainSocketImpl.java:350)
        at java.net.AbstractPlainSocketImpl.connectToAddress(AbstractPlainSocketImpl.java:206)
        at java.net.AbstractPlainSocketImpl.connect(AbstractPlainSocketImpl.java:188)
        at java.net.SocksSocketImpl.connect(SocksSocketImpl.java:392)
        at java.net.Socket.connect(Socket.java:589)
        at sun.net.NetworkClient.doConnect(NetworkClient.java:175)
        at sun.net.www.http.HttpClient.openServer(HttpClient.java:463)
        at sun.net.www.http.HttpClient.openServer(HttpClient.java:558)
        at sun.net.www.http.HttpClient.<init>(HttpClient.java:242)
        at sun.net.www.http.HttpClient.New(HttpClient.java:339)
        at sun.net.www.http.HttpClient.New(HttpClient.java:357)
        at sun.net.www.protocol.http.HttpURLConnection.getNewHttpClient(HttpURLConnection.java:1220)
        at sun.net.www.protocol.http.HttpURLConnection.plainConnect0(HttpURLConnection.java:1156)
        at sun.net.www.protocol.http.HttpURLConnection.plainConnect(HttpURLConnection.java:1050)
        at sun.net.www.protocol.http.HttpURLConnection.connect(HttpURLConnection.java:984)
        at com.getsentry.raven.connection.HttpConnection.doSend(HttpConnection.java:157)
        ... 5 common frames omitted

My questions is what am I doing wrong here.

When this issue has been solved I would love to contribute to an installation instruction for the Raspberry Pi on your website.

phrack, 2018-11-16:
The exceptions in the list a red herring. You're getting those because we no longer have a Sentry server, but ShootOFF will carry on anyway. The real issue is this:

`Error: Could not find or load main class com.shootoff.Main`

Couple questions:

1. Can you paste the command line you used in reply here?
2. Can you ensure the jar isn't corrupted? One way to do this is to extract it (JAR files are zip files with the extension .jar) and ensure the contents include com/shootoff/Main.class.

jtdaling, 2018-11-24:
I also tried java 7.

Now I got a different failure message:

java -jar ShootOFF.jar 
Exception in thread "main" java.lang.UnsupportedClassVersionError: com/shootoff/Main : Unsupported major.minor version 52.0
        at java.lang.ClassLoader.defineClass1(Native Method)
        at java.lang.ClassLoader.defineClass(ClassLoader.java:800)
        at java.security.SecureClassLoader.defineClass(SecureClassLoader.java:142)
        at java.net.URLClassLoader.defineClass(URLClassLoader.java:449)
        at java.net.URLClassLoader.access$100(URLClassLoader.java:71)
        at java.net.URLClassLoader$1.run(URLClassLoader.java:361)
        at java.net.URLClassLoader$1.run(URLClassLoader.java:355)
        at java.security.AccessController.doPrivileged(Native Method)
        at java.net.URLClassLoader.findClass(URLClassLoader.java:354)
        at java.lang.ClassLoader.loadClass(ClassLoader.java:425)
        at sun.misc.Launcher$AppClassLoader.loadClass(Launcher.java:308)
        at java.lang.ClassLoader.loadClass(ClassLoader.java:358)
        at sun.launcher.LauncherHelper.checkAndLoadMain(LauncherHelper.java:482)

phrack, 2018-11-24:
Yea, Java 7 won't work because ShootOFF uses code that was new in Java 8. Java 8 or 9 are the only real options, but Java 8 is all I've ever used.

Did you do the OpenCV step? The command wouldn't be `java -jar ShootOFF.jar`, it'd be something like: `java -Djava.library.path=$PATH:.:/usr/lib/jni/ -jar ShootOFF.jar --headless`. You need the headless part otherwise the GUI will start. If you do the headless part you need the Android app to get going.

phrack, 2018-11-24:
You'd have to read through two threads, but app stuff here:

https://www.reddit.com/r/ShootOFF/comments/5xjlaj/sbcapp_update/

If you want it, I'll create a repo with the Android app source code.

jtdaling, 2018-11-26:
My purpose is to permanently connecta raspberry pi to a beamer.
I would like to control to the raspberry via VNC.
The raspberry has a GUI installed.

This way I could make a fixed setup and I wouldn't have to manually install and calibrate the beamer each time I want to practice.

I guess I don't need the headless setup, I just need to find a solution to run ShootOff on a arm processor.
Is there any solution for that?

phrack, 2018-12-01:
You're going to run into a different problem: The pi's processor is not powerful enough to run shot detection comfortably. You'll end up with a very laggy GUI and missed shots. When we were working on this we were not planning to run shot detection on the pi, we had a camera to run that on (would only work with IR cartridges though). In the end the camera's vendor refused to support Linux or provide information to enable others to do so, and the driver we may was hacky and not suitable for regular use.

jtdaling, 2018-12-02:
Thanks for the info! I tried my idea out with a debian pc and ran into yet another problem:
you need a monitor besides a beamer to configure Shootoff. It was not possible to attach a beamer via VNC so I guess it was a bad idea after-all.

phrack, 2018-12-02:
I am sure you learned a lot in the process, so not a bad idea :).