phrack/ShootOFF #717: Failure to execute ShootOFF.jar with OpenJDK 1.8.0_252 on Ubuntu 20.04

HenryFBP, 2020-06-25:
I get `Error: Could not find or load main class com.shootoff.Main` when I execute `java -jar ShootOFF.jar` using OpenJDK 1.8.0_252.

```
durablepotato at jacks-imac-pro-x in ~/D/shootoff-4.0-final
» /usr/lib/jvm/java-1.8.0-openjdk-amd64//bin/java -jar ShootOFF.jar                                                                                                                     (base) 
Error: Could not find or load main class com.shootoff.Main
durablepotato at jacks-imac-pro-x in ~/D/shootoff-4.0-final
» cat /etc/os-release                                                                                                                                                                   (base) 
NAME="Ubuntu"
VERSION="20.04 LTS (Focal Fossa)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Ubuntu 20.04 LTS"
VERSION_ID="20.04"
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
VERSION_CODENAME=focal
UBUNTU_CODENAME=focal
```

flaflip, 2020-10-24:
Hi! same thing here.. Running it on a raspberry pi 4 with latest version of raspbian with openjdk 11.0.8. javafx is installed.
looks like ShootOff is not compatible with the latest version of Java...?

Should we downgrade java version :-( or do you have another fix?

nixthing, 2020-10-28:
I am using Ubuntu 20.04.
Had similar issue, after installing openjdk-8-jre had to do manual install the script under Linux install does not work at all.
Also did the recommended export LD_PRELOAD=`find /usr/lib -name "v4l1compat.so"`; java -jar ShootOFF.jar did not work, did some searching and so set Java environment in /usr/environment was finally able to launch but now get Error: **Could not find or load main class com.shootoff.Main.**
Can't go any further.

nixthing, 2020-10-29:
Okay solved my problem.
You have to use Oracle Java OpenJDK does not work.
Now have ShootOFF working.

nixthing, 2020-11-02:
Installed clean install 16.04. Definitely need Ubuntu 16.04 and Oracle Java 8. Don't know if Ubuntu 17, 18, 19 will work already shot (pun unintended) three days working on this. Maybe someday will test other variants of Ubuntu and Linux OS's but since they are for the most part essentially the same betting nothing recent will work.
However the problem still remains in Windows as well as Ubuntu 16.04 not picking up shots.
Tried setting camera in many different exposure settings, contrast, saturation, on white paper for target, camera  varying distances from target, shooting laser at varying distance to inches away and ShootOFF still not picking up shots placed on target. Although I can see the shoots on the console window. They are just not getting registered.
I am using Pink Rhino Laser ammo so ordered Laser-Ammo as that's what is shown being used in tutorials. Hope that does it.

dogweather, 2020-11-06:
> Could not find or load main class com.shootoff.Main

I get this as well with the current Oracle JDK download for Mac.

dogweather, 2020-11-06:
Doh - I didn't realize this project hasn't had a commit in five years.

nixthing, 2020-11-09:
True but it still works just need to use "older" OS, works just fine with Lubuntu 16.04 even in a VM, and in WIndows 10 so far. You also need to make sure you are using Oracle Java 8 and a good laser not a cheap one. I found Laser Ammo to be the best. SIRT should also work well.

whitis, 2022-01-01:
Lots of problems.
the export command will screw up every command you try to run afterwards.  Should remove the export and the semicolon or put the whole thing in parenthesis.  also, /usr/lib, like any directory passed to find, should have a trailing slash in case /usr/lib is a symlink.  
    LD_PRELOAD=`find /usr/lib -name "v4l1compat.so"` java -jar ShootOFF.jar
find: ‘/usr/lib/usbauth-notifier’: Permission denied
ERROR: ld.so: object '/usr/lib/i386-linux-gnu/libv4l/v4l1compat.so
/usr/lib/x86_64-linux-gnu/libv4l/v4l1compat.so' from LD_PRELOAD cannot be preloaded (cannot open shared object file): ignored.
Error: Could not find or load main class com.shootoff.Main
Caused by: java.lang.NoClassDefFoundError: javafx/application/Application

ls -lF /usr/lib/x86_64-linux-gnu/libv4l/v4* /usr/lib/i386-linux-gnu/libv4l/v4*
-rw-r--r-- 1 root root  5324 Feb 10  2018 /usr/lib/i386-linux-gnu/libv4l/v4l1compat.so
-rw-r--r-- 1 root root  5372 Feb 10  2018 /usr/lib/i386-linux-gnu/libv4l/v4l2convert.so
-rw-r--r-- 1 root root  5912 Feb 10  2018 /usr/lib/x86_64-linux-gnu/libv4l/v4l1compat.so
-rw-r--r-- 1 root root 10080 Feb 10  2018 /usr/lib/x86_64-linux-gnu/libv4l/v4l2convert.so

so we need to add "| tail -1" because we can only use one filename without proper punctuation.  I also replaced the outdated backticks:
LD_PRELOAD=$(find /usr/lib -name "v4l1compat.so" | tail -1) java -jar ShootOFF.jar

And we still have
Error: Could not find or load main class com.shootoff.Main
Caused by: java.lang.NoClassDefFoundError: javafx/application/Application

Which according to others means the program is unacceptably dependent on Oracle Java instead of OpenJDK.   Oracle Java's license has become totally unacceptable. It is no longer free of charge for most uses.   But even if we did try to install it, and manually select that version while everything else uses OpenJDK, it fails to install on Ubuntu 18.04.6 LTS, even though there is supposedly a 18.04 (bionic) package in the PPA.

sudo apt-add-repository ppa:webupd8team/java
sudo apt-get update
sudo apt-get install oracle-java8-installer
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Package oracle-java8-installer is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'oracle-java8-installer' has no installation candidate

looking back at what scrolled off the screen, apt-add-repository noted that java 9 has reached end of life.

https://www.linuxuprising.com/2021/09/how-to-install-oracle-java-17-lts-on.html

sudo add-apt-repository ppa:linuxuprising/java
sudo apt-get update
apt-cache search oracle-java
# no-install-recommends prevents making it the default jvm and making other program uses license violations.
sudo apt-get install --no-install-recommends oracle-java17-installer
# make sure oracle hasn't become default and see how to invoke oracle
update-alternatives --list java
   /usr/lib/jvm/java-11-openjdk-amd64/bin/java
   /usr/lib/jvm/java-17-oracle/bin/java
update-alternatives --display java
   java - auto mode
     link best version is /usr/lib/jvm/java-11-openjdk-amd64/bin/java
     link currently points to /usr/lib/jvm/java-11-openjdk-amd64/bin/java
     link java is /usr/bin/java
       slave java.1.gz is /usr/share/man/man1/java.1.gz
    /usr/lib/jvm/java-11-openjdk-amd64/bin/java - priority 1111
  slave java.1.gz: /usr/lib/jvm/java-11-openjdk-amd64/man/man1/java.1.gz
   /usr/lib/jvm/java-17-oracle/bin/java - priority 1091
   slave java.1.gz: /usr/lib/jvm/java-17-oracle/man/man1/java.1.gz

 LD_PRELOAD=$(find /usr/lib -name "v4l1compat.so" | tail -1) /usr/lib/jvm/java-17-oracle/bin/java -jar ShootOFF.jar

And after all that, it still can't find Main
find: ‘/usr/lib/usbauth-notifier’: Permission denied
Error: Could not find or load main class com.shootoff.Main
Caused by: java.lang.NoClassDefFoundError: javafx/application/Application

And if we try it with a fixed export (in parenthesis), it still doesn't work
(export LD_PRELOAD=$(find /usr/lib -name "v4l1compat.so" | tail -1); /usr/lib/jvm/java-17-oracle/bin/java -jar ShootOFF.jar)
find: ‘/usr/lib/usbauth-notifier’: Permission denied
Error: Could not find or load main class com.shootoff.Main
Caused by: java.lang.NoClassDefFoundError: javafx/application/Application

# based on what we see at https://stackoverflow.com/questions/56622327/error-could-not-find-or-load-main-class-application-main-javafx we try:
(export LD_PRELOAD=$(find /usr/lib -name "v4l1compat.so" | tail -1); /usr/lib/jvm/java-17-oracle/bin/java --module-path "/usr/share/java/" --add-modules javafx.controls,javafx.fxml -jar ShootOFF.jar)
find: ‘/usr/lib/usbauth-notifier’: Permission denied
Error occurred during initialization of boot layer
java.lang.module.FindException: Two versions of module maven3.embedder found in /usr/share/java (maven3-embedder-3.6.0.jar and maven3-embedder.jar)

So we try without the redundant module path:
(export LD_PRELOAD=$(find /usr/lib -name "v4l1compat.so" | tail -1); /usr/lib/jvm/java-17-oracle/bin/java --add-modules javafx.controls,javafx.fxml -jar ShootOFF.jar)
find: ‘/usr/lib/usbauth-notifier’: Permission denied
Error occurred during initialization of boot layer
java.lang.module.FindException: Module javafx.controls not found

 find /usr/lib/jvm/java-17-oracle/ -iname '*javafx*' 

So the program is still broken.

At least the jar file doesn't contain log4j (security hole)
zipgrep -i log4j2 ShootOFF.jar 
zipgrep -i JndiLookup.class ShootOFF.jar

germanicianus, 2022-08-27:
Currently I'm on Ubuntu 18.04.6 LTS and I got ShootOFF to run. In short for x64 architecture:
1. Download latest Java 8 from https://www.java.com/en/download/ (this still ships with JavaFX) and install or extract it.
2. Verify that you have the needed v4l library with `find /usr/lib -iname v4l1compat.so`.
3. Download ShootOFF, extract it and `cd` into the extracted directory.
4. Verify that ShootOFF launches with commandline (adopt v4l1compat.so path accordingly):
`LD_PRELOAD=/usr/lib/x86_64-linux-gnu/libv4l/v4l1compat.so <path-to-your-extracted-jre>/bin/java -jar ShootOFF.jar`
5. Quit ShootOFF because it didn't find your integrated/USB camera(s).
6. Set up your integrated/USB camera as IP camera using motion:
https://motion-project.github.io/motion_guide.html
7. Add the just set up IP camera to ShootOFF.
8. Have fun.

[image]

germanicianus, 2022-08-27:
Howto: ShootOFF on Ubuntu 18.0.6 LTS with currently latest x64 Adopt OpenJDK 16.0.2+7 and OpenJavaFX 18.0.2
1. Download and extract ShootOFF to a directory of your choice, e.g. `~/shootoff-4.0-final/`.
1. Download and extract Adopt OpenJDK to a directory of your choice, e.g. `~/java/adopt_openjdk-16.0.2+7/`.
https://adoptium.net/temurin/releases
2. Download and extract OpenJavaFX to a directory of your choice, e.g. `~/java/openjavafx-sdk-18.0.2/.`
https://gluonhq.com/products/javafx/
3. Extract file `~/shootoff-4.0-final/libs/opencv-2.4.13-0.jar/nu/pattern/opencv/linux/x86_64/libopencv_java2413.so` to directory `~/shootoff-4.0-final/`. Later in the command line, latter directory must be specified in the `java.library.path` because otherwise the library won't be found and ShootOFF won't start.
4. `cd` into `~/shootoff-4.0-final/` and launch ShootOFF with command line:
`LD_PRELOAD=/usr/lib/x86_64-linux-gnu/libv4l/v4l1compat.so ~/java/adopt_openjdk-16.0.2+7/bin/java --module-path ~/java/openjavafx-sdk-18.0.2/lib --add-modules javafx.controls,javafx.fxml --add-exports java.desktop/sun.awt.image=ALL-UNNAMED -Djava.library.path="./" -jar ShootOFF.jar`
You may also use:
`LD_PRELOAD=/usr/lib/x86_64-linux-gnu/libv4l/v4l1compat.so ~/java/adopt_openjdk-16.0.2+7/bin/java --module-path ~/java/openjavafx-sdk-18.0.2/lib --add-modules javafx.controls,javafx.fxml --add-exports java.desktop/sun.awt.image=ALL-UNNAMED -cp "ShootOFF.jar:./libs/*.jar" -Djava.library.path="./" com.shootoff.Main`

Valiant-Mouse, 2023-01-10:
The how to by @germanicianus works on Parrot OS 6.0.0 with javafx-sdk-19 and jdk-17.05+8, which are the latest versions from https://adoptium.net/temurin/releases and https://gluonhq.com/products/javafx/ ... I just had to update the filenames.