phrack/ShootOFF #719: CO2 Recoli shot not recognized

evazzoler, 2020-12-04:
I don't know if this is a ShootOFF issue or a system limit override. And I don't know if ShootOFF is an abandouned project or not.
It would be a real shame, as it is a beautiful software, better than many other paid ones.

I set up ShootOFF with a HD projector and a Logitech C920 camera on a dedicated (clean new) Windows 10 laptop (no way to make C920 recognizable on my loved Ubuntu 20.04).
I use it with a laser cartridge and all is good as expected.
When I shoot with a CO2 Umarex with vibration activated cartridge (see Laser Ammo products), the recoil produces a laser strip instad a laser point and ShootOFF dont'recognizes it as a shot. I can perfectly see that strip in realtime on the PC monitor in the arena webcam view, so is not correct to say that ShootOFF can't see it.
I think that ShootOFF should recognize laser strips (placing the shot where the strip starts) when a configuration based option is activated (i.e.: "recoil shot handgun mode")...
It would be the only way to shoot with a almost-real weapon.
Thank you for supporting.

sfraise, 2021-12-22:
I have a hunch this is due to the camera/detection speed.

Smokeless Range (laser ammo simulator) uses a high speed camera in their application for a reason.

I'm debating forking this Java version and tinkering around with it or just starting a new project trying to use something like tracking.js or opencv in an electron app.  It seems this project has gone dormmate and I'm thinking a more modern / easier to work with stack might be the better way to go if possible, I'm just not sure how long it will take me to develop part time as a side project.  Also, it's been a long long minute since I've done anything in Java lol.

I have both a laser cartridge and a spider vibration visible laser thing (assuming same exact thing as you) so I'll tinker around with it regardless of which way I go.

jankir123, 2024-02-05:
Did you ever manage to do something about this? I am looking at the exact same setup. Thanks!

sfraise, 2024-02-05:
> Did you ever manage to do something about this? I am looking at the exact same setup. Thanks!

I never got time to mess with it between work and buying a house to renovate it just went by the way.

It would be cool if someone would pick it up though and keep an open source free or at least freemium version going.

Camera speed is always going to be an issue though I think.