phrack/ShootOFF #720: Launching ShootOff on a 2018 MacBook Pro (Catalina 10.15.7) fails

ScottyIPSC, 2020-12-21:
Hi guys,
I unpacked your archive and tried to launch ShootOFF. it fails with the message "**The Java JAR file "shootOFF.jar" could not be launched.**"

Launching the diagnostics app reveals the following problem:

Runtime: Java(TM) SE Runtime Environment 15.0.1+9-18 64-bit
OS: Mac OS X 10.15.7 x86_64, Patch Level: null
Max amount of memory JVM will use = 4096 MB
Intel(R) Core(TM) i7-8559U CPU @ 2.70GHz
 is64bit = true
 physical cpu(s): 4
 logical cpu(s): 8
CPU Load Per Processor: 22,85% 3,28% 19,94% 3,24% 17,53% 3,25% 16,31% 3,27%
Total RAM Installed: 16384 MB
Available RAM: 8727 MB
ShootOFF Output:

exit code: 1

stdout:
---

stderr:
---
Error: Could not find or load main class com.shootoff.Main
Caused by: java.lang.NoClassDefFoundError: javafx/application/Application

Graceful shutdown took too much time, forcing the shutdown.

Any help to a newbie (me ;-)) is greatly appreciated!

Thanks a lot,
Alex Schott

milanek, 2020-12-21:
Hello Alex, you will have to downgrade your Java version to Java 8 - recent version does not contain necessary components.

ScottyIPSC, 2020-12-21:
Thanks so much milanek. You nailed it. Installing Java 8 allowed me to progress and to run into the next problem ;-)
It seems that Mac OS Catalina introduced an issue with granting access to cameras (no matter if built in or USB). 

Now the app terminates with the following message: 2020-12-21 14:11:00 [main] WARN  com.github.sarxos.webcam.Webcam - No webcam has been detected!

Any smart idea to fix that? 

Thanks again!
Alex