phrack/ShootOFF #721: Windows 10 projector issue

cfisher79, 2021-02-21:
I'm having starting the projector arena. I've watched the videos and read through the wiki and reddit and while I have seen some people having similar issues I can't find a solution. Shootoff starts up and runs fine my problem is when i get the arena window to come up that says needs calibration I can't drag and drop the arena window to the projector as it was shown in the video. With the projector connected I can through using the windows and p keys open a menu and mirror my screen and see the projector works but it doesn't open a screen of it's own. I have no place to drag the arena screen to for Shootoff recognize the projector and start the calibration process. Any help would be appreciated

MudMan54, 2021-12-13:
You need to use extended screens, not mirrored. Position the projector screen via the display properties to whatever position from the main screen.  Then drag the arena screen in that direction, it should then appear via the projector view.

cfisher79, 2021-12-26:
Yes I understand what needs to happen. As stated in the original post. There is no projector screen to position or drag and drop the arena screen to. How do I get the projector to open a screen so I can drag and drop the arena to it.